Sensirion/arduino-ble-gadget #1: Compile Errors

robo53, 2020-12-03:
Hi @MBjoern and @rnestler 
Thanks for the prompt replies.....
I tried the new code but your fix with `#include <cmath>` did not help....sorry, still the same errors.
`C:\Users\Robert Jirik\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\sdk\include\newlib\math.h`
I am still using still Win7/64bit.
Maybe you have an idea.
Many thanks in advance

MBjoern, 2020-12-03:
I've just pushed a promising fix for the mentioned issue... Release 0.4.1 will be available shortly.
Would you mind testing the current version, to see if it compiles without errors now?

MBjoern, 2020-12-03:
v0.4.1 should be available now... could you test if it resolves the mentioned issues?

robo53, 2020-12-03:
Hi @MBjoern 
Compiled bretty well. Only 1 uninteresting warning....
Thanks a lot and a good night from vienna.

closed 2020-12-04 by MBjoern