Sensirion/arduino-ble-gadget #2: Fetching data from app not working correctly

Nkae, 2021-02-08:
Hi! I'm new to Github!

I followed the tutorial to build the CO2 sensor with the sensirion SCD30, but modified this software to light up 3 LEDs depending on the CO2 level and have a rough idea on the air quality with a fast look.

The problem is that the Android app seems like it lost connection with the ESP, sometimes pressing the "fetch data" button fix it and the gaps in the plot are filled, but sometimes the download get stuck at 40~70%.

The  software was based on the Example2, modifications made:
- deleted the Serial.print for CO,Temp and RH values.
+ added a CO2_level variable where result[0] is stored.
+ added a switch case to handle the LEDs.

i'm not sure if it can be a problem caused by modifying this program or if this also happens as is and then it's an issue of the sensirion app.

I can upload the code if necessary. Thanks

MBjoern, 2021-02-08:
Hi Nkae

Thank you for reaching out to us.

To verify, whether your modifications have caused the download issue or if there is a bug in our source code, I suggest to run the used Example2 without any modifications and check, if you can reproduce the "Stuck download at 40-70%" bug with the original code.

Best regards
Björn

Nkae, 2021-02-08:
Hi MBjoern!

That was the next thing I had to try, I have set it up and will try to fetch data in a couple of ours.
I didn't mention in the first post but the logging interval is set to 10 seconds, because I was doing some test and wanted to have "real time" data, first tried with 2 seconds but took ages to download due to the high number of points and it also get stuck.

Sometimes if it fetch the data the plot shows a point that goes "back in time" as in this picture.
[image]

Nkae, 2021-02-08:
Same thing happen with unmodified example2, I was using V0.5, I can try with 0.7.1. Also the ESP was hooked to a laptop that went into sleep, now it's plugged to a wall adapter.

Nkae, 2021-02-09:
Yesterday I delete gadget from app, to start with new data. Download was ok but it left some gaps on the plot, and looking at the exported data:
 8 measurements at 17:21
21:49 to 23:21
1:40 to 2:20
4:41 to 5:23
7:36 until now.

This is still with example2 v0.5 plugged to a wall adapter and logging interval of 10 seconds.

MBjoern, 2021-02-09:
Hi Nkae

Thank you for your feedback... I'll look into it and will get back to you.

Best regards
Björn

Nkae, 2021-02-17:
Hi again!
I've trying during this week and it seems that downloading data from one sensor at a time is less prone to fail.
On sensors with 10 minutes interval the plots seems continuous but with 10 seconds plots looks like lots of data points in some parts and then a line until the next package of available data. This occurs even downloading data regularly.

I can upload data for the 10 seconds device if needed.

Nkae, 2021-03-24:
Hi MB!

I tried with another phone (samsung S9+) and at first it seems to work better when fetching data, but it fail too.
Here are 2 screenshots.
First image shows the point in the future at 0 ppm and back to the current time. I tried downloading data one by one but it failed, so i tried with the remaining sensors all at once, and failed.
Today I tried with another batch and one by one worked on the S9+ but failed on the first fetching on the other phone (Motorola Moto e6 play)

On the second one there are some sensors that have straight lines as if there was no data for that interval.

[image]

---

[image]

On both pictures 10 sensors where connected but only 8 where selected to plot.

MBjoern, 2021-03-29:
Hi there

Thank you very much for your detailed information on this matter. We'll look into it and release an updated version of the app soon. Please stay tuned and thank you for your patience.

Nkae, 2021-03-29:
Hi!
I have new info. I tried moving gadgetBle.handleEvents();  delay(3);, and I left them inside "if (esp_timer_get_time() - lastMmntTime >= startCheckingAfterUs) ".
I tried fetching data from 9 devices and all of them did well, really slow transfer, but all data was copied to the smartphone.
I tried with another SCD with the previous version with gadgetBLE.handleEvents in the main loop outside any other structure and failed.

Hope this also helps.

MBjoern, 2021-03-29:
Hi

What you could do, is increase the delay after handleEvents... this tells the BLE lib how fast to update the transmitted data. I wouldnät move it into the mmnt timer part, as you have tried, as this will end up being very slow... But you could try values bigger than 3 ms, like 7 -10 for a starter... 

Thank you very much for the update. This is highly appreciated.

Nkae, 2021-03-29:
I put it back out to where it was, it was a test. I'll try changing the delay on some ESPs.
Thank you!

Nkae, 2021-04-05:
> I put it back out to where it was, it was a test. I'll try changing the delay on some ESPs.
> Thank you!

With a 7 ms delay it has worked without any problem. I will keep trying and fetching data and update if it fails again.

MBjoern, 2021-04-06:
> With a 7 ms delay it has worked without any problem. I will keep trying and fetching data and update if it fails again.

That's great feedback, thank you for your tests. We'll consider updating the delay in the future as well, to support more types of devices.

Nkae, 2021-05-11:
Hi MB!
I think I didn't get this error since the update, but sometimes it get stuck when connecting or just after connecting at 0% download. I have to try with the default 3ms delay as it is 7ms now in my code.

If the device is a bit far with a weaker signal it may skip some points on the plots (also when downloading the data it jumps forward like a 15%),but getting close and fetching again usually fills the graph, other times a 3rd or 4th fetch is needed.

psachs, 2022-07-29:
Due to inactivity I will close this issue. We are trying to constantly improve the library. Let us know if you run into more issues

closed 2022-07-29 by psachs