Sensirion/arduino-ble-gadget #4: SCD30 not initializing

Enoch23, 2021-06-06:
Hello,

I have successfully compiled and uploaded the Example 7 to my esp32 board. I am using the M5Stick-C. I also edited the code of the example to match the pins I am using (32 and 33). The iOS application finds the sensor, however, because the sensor is not initialised, no data is shown.
The serial output is the following:
`Sensirion GadgetBle Lib initialized with deviceId = 7c:ca`
This is a screenshot of the application:
[image]

I know the sensor is not initialized because the light is not pulsing. I have been using this sensor with the library from Sparkfun:
https://github.com/sparkfun/SparkFun_SCD30_Arduino_Library
without issues, so I know the sensor works.

The edited Example 7 looks like this:

`#include "esp_timer.h"
#include <Wire.h>

// Download the SeeedStudio SCD30 Arduino driver here:
//  => https://github.com/Seeed-Studio/Seeed_SCD30/releases/latest
#include "SCD30.h"

#include "Sensirion_GadgetBle_Lib.h"

static int64_t lastMmntTime = 0;
static int startCheckingAfterUs = 1900000;
#define I2C_SDA 33 // added this line
#define I2C_SCL 32 // and this line
GadgetBle gadgetBle = GadgetBle(GadgetBle::DataType::T_RH_CO2);

void setup() {
  Serial.begin(115200);
  delay(100);

  // Initialize the GadgetBle Library
  gadgetBle.begin();
  Serial.print("Sensirion GadgetBle Lib initialized with deviceId = ");
  Serial.println(gadgetBle.getDeviceIdString());

  // Initialize the SCD30 driver
  Wire.begin(I2C_SDA,I2C_SCL); //Edited this line
  scd30.initialize();
  scd30.setAutoSelfCalibration(1);
  scd30.setTemperatureOffset(3);
}

void loop() {
  float result[3] = {0};

  if (esp_timer_get_time() - lastMmntTime >= startCheckingAfterUs) {

    if (scd30.isAvailable()) {
      scd30.getCarbonDioxideConcentration(result);

      gadgetBle.writeCO2(result[0]);
      gadgetBle.writeTemperature(result[1]);
      gadgetBle.writeHumidity(result[2]);

      gadgetBle.commit();
      lastMmntTime = esp_timer_get_time();

      // Provide the sensor values for Tools -> Serial Monitor or Serial Plotter
      Serial.print("CO2[ppm]:");
      Serial.print(result[0]);
      Serial.print("\t");
      Serial.print("Temperature[℃]:");
      Serial.print(result[1]);
      Serial.print("\t");
      Serial.print("Humidity[%]:");
      Serial.println(result[2]);
    }
  }

  gadgetBle.handleEvents();
  delay(3);
}`

MBjoern, 2021-06-07:
Hi Enoch23

Thank you for reaching out to us.

I've never tried this example with an M5Stick-C. We usually work with ESP32 DevKit V4, equipped with an ESP32-WROOM-32D.

I've double checked our Example7 from the most recent Library version available via Arduino Library Manager and everything worked like a charm, when using the wiring documented here: [SCD30 Tutorial](https://github.com/Sensirion/arduino-ble-gadget/blob/master/documents/SCD30_BLE_Gadget_with_RHT_Tutorial.md).

The Serial Monitor should provide you with the following output, if everything worked fine:

> Sensirion GadgetBle Lib initialized with deviceId = dd:26
> CO2[ppm]:874.54	Temperature[℃]:28.22	Humidity[%]:49.34
> CO2[ppm]:874.39	Temperature[℃]:28.15	Humidity[%]:49.28
> CO2[ppm]:873.87	Temperature[℃]:28.18	Humidity[%]:49.20
> CO2[ppm]:874.11	Temperature[℃]:28.16	Humidity[%]:49.11

As you've probably noticed, We're using the Seeed Studio SCD30 driver for our example, as we have not developed our own arduino driver for the SCD30. Have you tried just running an example from that library, to check if the driver might not be compatible with your hardware? As you've mentioned, the SCD30 is not initialized properly, so this might be the reason.

On the other hand, you've mentioned that it works fine with the SparkFun driver. As adding our BLE Gadget library to a given driver does not require lots of code, you can feel free to switch out the driver and apply our library to the SparkFunDriver isntead.

From SparkFun SCD30 Arduino Library's Example "Basic Reading", I breifly adapted the example to use our BLE library. Please see the changed lines maked with `// Sensirion` below. Give this a try and see if it works now. Don't forget to check your wiring and the pins used by SparkFun's driver.

    /*
      Reading CO2, humidity and temperature from the SCD30
      By: Nathan Seidle
      SparkFun Electronics
      Date: May 22nd, 2018
      License: MIT. See license file for more information but you can
      basically do whatever you want with this code.
    
      Feel like supporting open source hardware?
      Buy a board from SparkFun! https://www.sparkfun.com/products/15112
    
      This example prints the current CO2 level, relative humidity, and temperature in C.
    
      Hardware Connections:
      Attach RedBoard to computer using a USB cable.
      Connect SCD30 to RedBoard using Qwiic cable.
      Open Serial Monitor at 115200 baud.
    */
    
    #include <Wire.h>
    
    #include "SparkFun_SCD30_Arduino_Library.h" //Click here to get the library: http://librarymanager/All#SparkFun_SCD30
    SCD30 airSensor;
    
    // Sensirion vvv
    #include "Sensirion_GadgetBle_Lib.h"
    
    static int64_t lastMmntTime = 0;
    static int startCheckingAfterUs = 1900000;
    
    GadgetBle gadgetBle = GadgetBle(GadgetBle::DataType::T_RH_CO2);
    // Sensirion ^^^
    
    void setup()
    {
      Serial.begin(115200);
      Serial.println("SCD30 Example");
      Wire.begin();
    
      if (airSensor.begin() == false)
      {
        Serial.println("Air sensor not detected. Please check wiring. Freezing...");
        while (1)
          ;
      }
    
      //The SCD30 has data ready every two seconds
    
      // Sensirion vvv
      // Initialize the GadgetBle Library
      gadgetBle.begin();
      Serial.print("Sensirion GadgetBle Lib initialized with deviceId = ");
      Serial.println(gadgetBle.getDeviceIdString());
      // Sensirion ^^^
    }
    
    void loop()
    {
      // Sensirion vvv
      if (esp_timer_get_time() - lastMmntTime >= startCheckingAfterUs) {
        // Sensirion ^^^
        if (airSensor.dataAvailable())
        {
          Serial.print("co2(ppm):");
          Serial.print(airSensor.getCO2());
    
          Serial.print(" temp(C):");
          Serial.print(airSensor.getTemperature(), 1);
    
          Serial.print(" humidity(%):");
          Serial.print(airSensor.getHumidity(), 1);
    
          Serial.println();
    
          // Sensirion vvv
          gadgetBle.writeCO2(airSensor.getCO2());
          gadgetBle.writeTemperature(airSensor.getTemperature());
          gadgetBle.writeHumidity(airSensor.getHumidity());
    
          gadgetBle.commit();
    
          lastMmntTime = esp_timer_get_time();
          // Sensirion ^^^
    
        }
        else
          Serial.println("Waiting for new data");
      }
    
      // Sensirion vvv
      gadgetBle.handleEvents();
      delay(3);
      // delay(500); Remove this
      // Sensirion ^^^
    }

Please let me know if this resolves the issues mentioned.

Good Luck
Best regards
Björn

Enoch23, 2021-06-07:
!!!! Thank you so much !!!!
This example code is what I used for my project, so I knew it should work. It didn't however. So I tried uploading my previous code. That's when I noticed that the Library for the M5Stick-C Plus I am using was not installed. So I installed it and my previous code worked right away. That got me thinking what was different about my code to your modified example, since mine was also modified from that same example. Turns out, what was missing was an `#include <M5StickCPlus.h>` at the appropriate place and an `M5.begin();` within the `void setup() {}` part. That got your modified example working! I tried the exact same modifications to the Example 7 offered here and it worked as well!

So, again, thank you!

Here is the modified Example 7 for the M5Stick-C Pro using the Grove port:

`//#include <SparkFun_SCD30_Arduino_Library.h>
//SCD30 sensor
#include "esp_timer.h"
#include <Wire.h>
// M5 vvv
#include <M5StickCPlus.h>
//M5 ^^^

// Download the SeeedStudio SCD30 Arduino driver here:
//  => https://github.com/Seeed-Studio/Seeed_SCD30/releases/latest
#include "SCD30.h"

#include "Sensirion_GadgetBle_Lib.h"

static int64_t lastMmntTime = 0;
static int startCheckingAfterUs = 1900000;
// M5 vvv
#define I2C_SDA 33
#define I2C_SCL 32
//M5 ^^^
GadgetBle gadgetBle = GadgetBle(GadgetBle::DataType::T_RH_CO2);

void setup() {
// M5 vvv
  M5.begin();
// M5 ^^^
  Serial.begin(115200);
  delay(100);

  // Initialize the GadgetBle Library
  gadgetBle.begin();
  Serial.print("Sensirion GadgetBle Lib initialized with deviceId = ");
  Serial.println(gadgetBle.getDeviceIdString());

  // Initialize the SCD30 driver
  Wire.begin(I2C_SDA,I2C_SCL);
  scd30.initialize();
  scd30.setAutoSelfCalibration(1);
  scd30.setTemperatureOffset(3);
}

void loop() {
  float result[3] = {0};

  if (esp_timer_get_time() - lastMmntTime >= startCheckingAfterUs) {

    if (scd30.isAvailable()) {
      scd30.getCarbonDioxideConcentration(result);

      gadgetBle.writeCO2(result[0]);
      gadgetBle.writeTemperature(result[1]);
      gadgetBle.writeHumidity(result[2]);

      gadgetBle.commit();
      lastMmntTime = esp_timer_get_time();

      // Provide the sensor values for Tools -> Serial Monitor or Serial Plotter
      Serial.print("CO2[ppm]:");
      Serial.print(result[0]);
      Serial.print("\t");
      Serial.print("Temperature[℃]:");
      Serial.print(result[1]);
      Serial.print("\t");
      Serial.print("Humidity[%]:");
      Serial.println(result[2]);
    }
  }

  gadgetBle.handleEvents();
  delay(3);
}`

Could you please let my know how to format the code in this editor?

MBjoern, 2021-06-08:
I'm glad everything worked out. Thanks for your inputs regarding the handling of the M5 boards. That's good input for potential future customer requests.

If you add code in this editor: Select the code and press the Tab key on your keyboard to indent it. This will tell the editor to format it differently. Press on Preview, to see if it turned out as expected.
I hope that helps.

Keep up the good work!
Best regards 
Björn

closed 2021-06-08 by MBjoern