Sensirion/arduino-ble-gadget #5: Send Battery Level to Application

Enoch23, 2021-07-30:
Is it possible to send a battery level of the device to the application?

I was thinking it would be something of the sort like:
`gadgetBle.writeBattery(batteryLevel);`
but I cant find anything in the "Sensirion_GadgetBle_Lib.h". Is this feature implemented?

The SHT31 smart gadget was able to do so according to this post:
https://www.soselectronic.com/articles/sos-supplier-of-solution/temperature-measurement-with-sht31-smart-gadget-2330

P.S. I hope this is the correct way of asking. Or how does one go about feature requests here on GitHub?

MBjoern, 2021-07-30:
Hi Enoch23

Thank you for your suggestion.
Yes, the Smart Gadgets do have a Battery Service to provide battery level information. The arduino-ble-gadget library on the other hand does not provide this functionality (yet).

I'll add this to our backlog as a future improvement.

Best regards
Bjoern

P.S. Yes, this is the right channel and way to ask about stuff like this.

psachs, 2022-10-14:
You are correct that we currently will not support any battery service in MyAmbience and therefore we will not add it to the ble gadget library yet.

We will definitely consider this in a future update and add the service here as well as soon as it is supported.