Sensirion/arduino-ble-gadget #7: Question: (how) can I set the product type

marcelstoer, 2021-09-08:
In the gadget settings on the mobile I see "Product type: DIY Gadget". (How) can I give my fleet of devices a dedicated type? I didn't find this in the API and failed to figure out which internal value defines this.

MBjoern, 2021-09-09:
This is hardcoded in the MyAmbience app, to distinguish official sensirion gadgets and reference designs from Do-It-Yourself gadgets built with this library. Hence, there's no way to change this in the current version, but it's a good idea to allow some sort of product type characteristic configurable in the ble library.

marcelstoer, 2021-09-09:
Disclaimer: I'm one of the ThingPulse co-founders i.e. these guys here: https://www.linkedin.com/posts/sensirion_makermonday-makeitwithsensirion-indoorairquality-activity-6840644898165719040-I1jz/

As a device manufacturer we'd of course be interested in seeing our devices advertised in the mobile app under our product name. Shipping devices that have the "DIY" term attached to them in your app isn't exactly desirable 😉 even though it's a tiny detail.

Maybe you could allow defining the product name through the library and announced over BLE. In addition Sensirion could keep maintaining a repository of "official", "approved" or "endorsed" devices and state in the mobile app whether the product is official or not.

MBjoern, 2021-09-09:
> As a device manufacturer we'd of course be interested in seeing our devices advertised in the mobile app under our product name. Shipping devices that have the "DIY" term attached to them in your app isn't exactly desirable wink even though it's a tiny detail.
Yes, I totally understand this, and fully agree. Let me give this some thought about how we shall add this. Naming it "DIY Gadget" in this case is not quite optimal.

But keep in mind, that this library and the MyAmbience app's are designed for show case and inspirational purposes. Using them in commercial products flatters us, but will probably not get the support expected, when it comes to feature requests and bug reports.

marcelstoer, 2021-09-09:
> But keep in mind, that this library and the MyAmbience app's are designed for show case and inspirational purposes. 

Ok, sorry, didn't know that.

> ...will probably not get the support expected, when it comes to feature requests and bug reports

No worries, we're quite happy with what we can do with MyAmbience out-of-the-box.

closed 2021-09-09 by marcelstoer