Sensirion/arduino-ble-gadget #8: Question: why no configurable device id

marcelstoer, 2021-09-08:
What is the technical reason you couldn't allow opening up the device id to be manually defined? If I'm shipping devices to customers it could be desirable to give devices more "meaningful" identifiers than those derived from the BLE device address i.e. "CO2 monitor, room H67" rather than "C5:43".

MBjoern, 2021-09-09:
Hi

The device id is used as a unique identifier, to distinguish different devices. It should be reasonably unique. Hence, we use the parts of the BLE ma address to create this uniqueness (it's not really unique, as the id is quite short, but assuming the amount of devices available to a user is rather small, it's unique enough).

Now having more meaningful ids, such as "CO2 monitor, room H67" is a desirable feature, I totally agree. But the device-id is not suitable for this purpose, as the BLE protocol does not allow us to have big enough advertisement packets to store this amount of information. (The advertisement packets are data frames use to advertise the presenece of BLE devices near you.)

We currently save a matching of device-ids to "Gadget Names" on the MyAmbience app, but plan to extend this to allow to either store Gadget Names in the gadget itself (requires a connection to fetch the device name) or have it distributed in the app itself or via cloud download.

Let us know, if you have any ideas on this or feel free to open a Pull Request.

marcelstoer, 2021-09-09:
Appreciate your feedback, thanks.

> the BLE protocol does not allow us to have big enough advertisement packets to store this amount of information.

Alright, totally forgot about this constraint. Decoupling "device id" from "gadget name" (i.e. have two distinct properties for them) is probably a sane approach.

- Device id is something technical with constraints imposed by the BL(E) spec. The current implementation makes sense AFAIU.
- Device/gadget name is non-technical in that it could be some arbitrary text. The question here is: who defines it? IMO it's clear that it should not be defined by the device manufacturer/distributor but by the actual user(s). This in turn means it's rather a topic for the mobile app than for this library -> I'll close this issue.

closed 2021-09-09 by marcelstoer

marcelstoer, 2021-09-10:
Duuhh...it took me a while to figure out that the "Gadget Name" is actually _editable_ in the MyAmbience app. Sorry for the noise in this issue.

MBjoern, 2021-09-10:
No worries... Let me know, if you have any further feedback.