Sensirion/arduino-ble-gadget #10: Different PM sizes DataType

Nkae, 2021-11-12:
Hello MBjoern!

As you know I'm working with the SPS30 now, and I was wondering why there is no DataType for different PM sizes. What I do now is I use the T_RH_VOC_PM to test it with MyAmbience app.

As the SPS30 can detect different PM sizes, I think it would be a good thing to create a  PM1, PM2.5, PM4 and PM10 DataType (even the 2 last are calculated from PM2.5 values)

MBjoern, 2021-11-12:
Hi Nkae

Thank you for reaching out to me, and I'm glad that you've found a workaround in the mean time.
Hmmm, good point, I'll open up a ticket to tackle this enhancement at some point. No promises, but i like the idea.

Best regards
Björn

psachs, 2022-06-03:
The feature will be added with the 0.14.0 release within the next few days.

closed 2022-06-03 by psachs