Sensirion/arduino-ble-gadget #12: SensirionSimple25pt7b

zorg2015, 2022-01-22:
Hello,
I'am a beginner. So excuse me if my issue seems obvious for you.
I try to make a CO2 sensor based on a lilygo TTGO ESP32 board. I use the Arduino IDE. I have loaded the library zip file but when I verify the Example3_SCD30_BLE_TTGO_Gadget.ino file I have the following error : 
'SensirionSimple25pt7b' was not declared in this scope

I don't understand how to solve this issue

Thanks for your help

psachs, 2022-01-26:
Please follow the setup guide as described here:

https://github.com/Sensirion/arduino-ble-gadget/blob/master/documents/SCD30_Monitor_Tutorial.md#additional-steps-for-lilygo-ttgo-t-display-esp32

You have to download the TFT eSPI library and patch it correctly. If you donwload the ZIP make sure to  only copy the TFT_eSPI folder since this includes the library. Then you have to add all the header and fonts to the correct place within this library.

I hope this helps you setup the project properly

zorg2015, 2022-01-26:
Thanks Pascal for your help and time. After having followed your guidelines, I just met a compilation error "'round' is not a member of 'std'". I just replaced std:: round by round and then I succeeded to download the program and my new CO2 sensor works fine !
Thanks a lot !

closed 2022-01-26 by zorg2015