Sensirion/arduino-ble-gadget #15: Feature request

melkati, 2022-02-11:
Hi.

I see you are actively working these days on migrating to NimBLE . Thank you!

I have some feature requests for the next version. I hope you can include some of them.

1. To make public _sampleIntervalMs or publish a method to change it, so the application using the library can choose the logging interval on init (right now it starts hardcoded at 600000, 1 minute, and user must go into the App after each firmware restart to set again their preferred sample interval). 

2. To be able to change SAMPLE_BUFFER_SIZE_BYTES dynamically via a method (at least on init) so it can be adapted to different use cases.

3. To make the _sampleBuffer public, or include some methods to read it, so this information can be used to, for example, make a chart with the logged data.  I would like to implement some charts on [my CO2 meter project ](https://github.com/melkati/CO2-Gadget) but right now do it I need to duplicate this data in my own buffer (a waste of memory).

Question: Is it possible to init the library (GadgetBle gadgetBle = GadgetBle(GadgetBle::DataType::T_RH_CO2_ALT)) in a mode that the App shows just CO2? It looks like the App is capable of it but can't find how to do it. 
I need this option to "hide" temperature and humidity readings when the sensor is mounted in a very constrained space, so readings are affected by heat emitted by the surrounding components.
[image]

Thank you very much and, please, **keep working on this great project**!

psachs, 2022-03-28:
Hi @melkati 

Sorry for the late answer. We are currently refactoring the library to provide better interfaces and architecture. Thank you very much for your input. I will check if we can support your requests in a future version.

1.) Yes this is a good point. In the app we only support a specific set of logging interval. To not break the app we would also need to limit the selection on supported logging intervals.

2.) The buffer is allocated on the stack, so there is no easy way to dynamically change this size without changing the library code.

3.) The sample buffer is a ring buffer used as an internal data structure for the history download feature. I have to check if we can provide a reliable interface to get the data for local plotting.

Currently the library only supports a data type for CO2 which also includes relative humidity and temperature. The RH/T values are removed by the App based on the gadget type. Unfortunately to support this at the moment a couple of changes to the library are required in the header:

```c++
static const char* const GADGET_NAME = "MyCO2";
static const size_t MAX_SAMPLE_SIZE = 8;
```

I will see if we can support an additional data type that only reports CO2 values in a future version

psachs, 2022-07-29:
Hi @melkati 

Small update. BLE Gadget library 0.14.0 or later supports a new data type to only report CO2 values (without humidity and temperature)

closed 2022-10-14 by psachs