Sensirion/arduino-ble-gadget #18: Trying to compile for Adafruit ESP32-S2 TFT Feather

indymx, 2022-07-20:
Getting the following errors. Any ideas? 

In file included from C:\Users\Gary Terry\AppData\Local\Temp\.arduinoIDE-unsaved2022619-19088-q71n9d.8y71d\Example10_SEN54_BLE_Gadget\Example10_SEN54_BLE_Gadget.ino:10:
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:48:44: error: expected class-name before ',' token
 class GadgetBle: BLECharacteristicCallbacks, BLEServerCallbacks {
                                            ^
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:48:65: error: expected class-name before '{' token
 class GadgetBle: BLECharacteristicCallbacks, BLEServerCallbacks {
                                                                 ^
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:81:14: error: 'map' in namespace 'std' does not name a template type
         std::map<Unit, UnitEnc> unitEnc;
              ^~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:81:9: note: 'std::map' is defined in header '<map>'; did you forget to '#include <map>'?
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:18:1:
+#include <map>
 
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:81:9:
         std::map<Unit, UnitEnc> unitEnc;
         ^~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:130:5: error: 'BLEAdvertising' does not name a type
     BLEAdvertising* _bleAdvertising;
     ^~~~~~~~~~~~~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:131:5: error: 'BLE2902' does not name a type
     BLE2902* _transferDescr;
     ^~~~~~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:132:5: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
     BLECharacteristic* _transferChar;
     ^~~~~~~~~~~~~~~~~
     Character_h
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:133:5: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
     BLECharacteristic* _sampleCntChar;
     ^~~~~~~~~~~~~~~~~
     Character_h
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:134:5: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
     BLECharacteristic* _wifiSsidChar;
     ^~~~~~~~~~~~~~~~~
     Character_h
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:177:20: error: 'BLEServer' has not been declared
     void onConnect(BLEServer* serverInst);
                    ^~~~~~~~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:178:23: error: 'BLEServer' has not been declared
     void onDisconnect(BLEServer* serverInst);
                       ^~~~~~~~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:181:18: error: 'BLECharacteristic' has not been declared
     void onWrite(BLECharacteristic* characteristic);
                  ^~~~~~~~~~~~~~~~~

indymx, 2022-07-28:
Anyone have any ideas how to fix this issue??

psachs, 2022-07-29:
Hi @indymx 

To me it looks like you are missing a dependency that leads to errors.

As you are using the SEN54 example please make sure you also install the "Sensirion I2C SEN5X" library through the Arduino Library Manager.

I assume you installed the BLE Gadget Library through the library manager as well. I just tested the compilation of the example in Arduino IDE 1.8.13 and it worked for me without any errors or warnings.

indymx, 2022-07-29:
I have both libraries installed from the library manager..

[image]

indymx, 2022-07-29:
I believe I found a possible source of the issue. Seems that the Adafruit ESP32_S2 boards don't have the esp_timer.h packaged with them... need to figure out how to fix that problem..

psachs, 2022-07-29:
It's hard for me to track down the cause of the error. Since you have installed all the libraries I suspect that something is incompatible with either a dependency or the compiler.
Can you provide me the version of you Arduino IDE and the ESP32 Arduino library you have installed? I can then try to reproduce the problem.

psachs, 2022-07-29:
Btw we just did a major update to the library and removed the dependency on esp_timer.h in favor of the built in Arduino timer.

I tested compilation with the "Adafruit ESP32 Feather" board and this worked for me, so I guess I used the wrong board configuration for testing

indymx, 2022-07-29:
Is that library version available to the public yet?

indymx, 2022-07-29:
> It's hard for me to track down the cause of the error. Since you have installed all the libraries I suspect that something is incompatible with either a dependency or the compiler. Can you provide [...]

I am using Arduino 2.0.0_RC9 / ESP32 lib is 2.0.2, just now installing 2.0.4 and will test with that

indymx, 2022-07-29:
no change with the 2.0.4 ESP 32.

psachs, 2022-07-29:
@indymx 

Yes its Version 1.0.0 and publicly available.

We changed to NimBLE which needs to be installed first and the API changed quit a bit. Please refer to the examples or the migration guide if you already wrote your own code.

Best regards.

indymx, 2022-07-29:
I am using the example code provided with the library, unmodified.. 

it still has the include for esp_timer. 

Documentation does not state that NimBLE needs to be installed, may want to address that. 

/*
 GadgetBle.h - Library for providing sensor data via Bluetooth.
 Created by Bjoern Muntwyler, November 7, 2020.
 Released under BSD-3 licence
 */
#ifndef Sensirion_GadgetBle_Lib_h
#define Sensirion_GadgetBle_Lib_h

#include "Arduino.h"
#include <functional>

#include "esp_timer.h"

#include <BLE2902.h>
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>

indymx, 2022-07-29:
with NimBLE installed, no change

psachs, 2022-07-29:
Yes, I can see it is still used in the and needs to be changed to fix the compilation issue.

As a workaround you can replace the esp_timer with the millis() function from Arduino as in the patch posted below.
We will fix this issue in the next version

```
diff --git a/src/Sensirion_GadgetBle_Lib.cpp b/src/Sensirion_GadgetBle_Lib.cpp
index 7958045..ea3b843 100644
--- a/src/Sensirion_GadgetBle_Lib.cpp
+++ b/src/Sensirion_GadgetBle_Lib.cpp
@@ -255,8 +255,8 @@ void GadgetBle::writeHCHO(float value) {
 }
 
 void GadgetBle::commit() {
-    if (esp_timer_get_time() - _lastCacheTime >= (_sampleIntervalMs * 1000)) {
-        _lastCacheTime = esp_timer_get_time();
+    if (millis() - _lastCacheTime >= _sampleIntervalMs) {
+        _lastCacheTime = millis();
         _addCurrentSampleToHistory();
     }
 
diff --git a/src/Sensirion_GadgetBle_Lib.h b/src/Sensirion_GadgetBle_Lib.h
index 3449f24..f986873 100644
--- a/src/Sensirion_GadgetBle_Lib.h
+++ b/src/Sensirion_GadgetBle_Lib.h
@@ -9,8 +9,6 @@
 #include "Arduino.h"
 #include <functional>
 
-#include "esp_timer.h"
-
 #include <BLE2902.h>
 #include <BLEDevice.h>
 #include <BLEServer.h>
```

indymx, 2022-07-29:
I applied the above patch, but still get this error:

`
In file included from C:\Users\Gary Terry\AppData\Local\Temp\.arduinoIDE-unsaved2022629-16804-1vom9sp.c3s3\Example10_SEN54_BLE_Gadget\Example10_SEN54_BLE_Gadget.ino:9:
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:47:44: error: expected class-name before ',' token
 class GadgetBle: BLECharacteristicCallbacks, BLEServerCallbacks {
                                            ^
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:47:65: error: expected class-name before '{' token
 class GadgetBle: BLECharacteristicCallbacks, BLEServerCallbacks {
                                                                 ^
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:80:14: error: 'map' in namespace 'std' does not name a template type
         std::map<Unit, UnitEnc> unitEnc;
              ^~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:80:9: note: 'std::map' is defined in header '<map>'; did you forget to '#include <map>'?
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:17:1:
+#include <map>
 
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:80:9:
         std::map<Unit, UnitEnc> unitEnc;
         ^~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:129:5: error: 'BLEAdvertising' does not name a type
     BLEAdvertising* _bleAdvertising;
     ^~~~~~~~~~~~~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:130:5: error: 'BLE2902' does not name a type
     BLE2902* _transferDescr;
     ^~~~~~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:131:5: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
     BLECharacteristic* _transferChar;
     ^~~~~~~~~~~~~~~~~
     Character_h
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:132:5: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
     BLECharacteristic* _sampleCntChar;
     ^~~~~~~~~~~~~~~~~
     Character_h
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:133:5: error: 'BLECharacteristic' does not name a type; did you mean 'Character_h'?
     BLECharacteristic* _wifiSsidChar;
     ^~~~~~~~~~~~~~~~~
     Character_h
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:176:20: error: 'BLEServer' has not been declared
     void onConnect(BLEServer* serverInst);
                    ^~~~~~~~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:177:23: error: 'BLEServer' has not been declared
     void onDisconnect(BLEServer* serverInst);
                       ^~~~~~~~~
c:\Users\Gary Terry\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src/Sensirion_GadgetBle_Lib.h:180:18: error: 'BLECharacteristic' has not been declared
     void onWrite(BLECharacteristic* characteristic);
                  ^~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
`

psachs, 2022-07-29:
I had to update the espressif board library to get the right dependency. It looks like you compiler might not show a useful error message. When I try to compile fo Adafruit Feather ESP32-S2 I get the following error.

```
In file included from /home/psachs/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/NimBLELibraryWrapper.cpp:2:
/home/psachs/Arduino/libraries/NimBLE-Arduino/src/NimBLEDevice.h:46:12: fatal error: esp_bt.h: No such file or directory
 #  include "esp_bt.h"
            ^~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Adafruit Feather ESP32-S2 TFT.
```

After a little research I found out that this ESP32 board has no Bluetooth support which is required for this library to work.

https://learn.adafruit.com/adafruit-esp32-s2-feather
> Please note the Feather ESP32-S2 has a single-core 240 MHz chip, so it won't be as fast as ESP32's with dual-core. Also, there is no Bluetooth support.

indymx, 2022-07-29:
SMH!.. I guess I should have checked that first. Now I just feel dumb.. ;)

No worries, I have 2 of these on the way. https://www.adafruit.com/product/5395 

They definitely have BLE support and are the correct form factor for what I am trying to build..

psachs, 2022-07-29:
No worry,

We mainly tested this library with the ESP32 Devkit and ESP32 WROVER E devkit which are very powerful and support a lot of features.

Since we found the root cause, I will close this issue. Let us know if you can successfully compile the code for the new boards that are on the way.

closed 2022-07-29 by psachs