Sensirion/arduino-ble-gadget #20: not compiled "Example3_SCD30_BLE_TTGO_Gadget"

RatsM, 2022-08-04:
https://github.com/Sensirion/arduino-ble-gadget/blob/0a0feb7760c48967471ed08af868b2da335e3bf0/examples/Example3_SCD30_BLE_TTGO_Gadget/Example3_SCD30_BLE_TTGO_Gadget.ino#L149

Example3_SCD30_BLE_TTGO_Gadget.ino: In function 'void loop()':
Example3_SCD30_BLE_TTGO_Gadget:149:29: error: 'round' is not a member of 'std'
       displayCo2((uint16_t) std::round(result[0]));
                             ^
C:\Users\RatsM\Documents\Arduino\Example3_SCD30_BLE_TTGO_Gadget\Example3_SCD30_BLE_TTGO_Gadget.ino:149:29: note: suggested alternative:
In file included from C:\Users\RatsM\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal.h:34:0,
                 from C:\Users\RatsM\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:35,
                 from sketch\Example3_SCD30_BLE_TTGO_Gadget.ino.cpp:1:
C:\Users\RatsM\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
               ^
exit status 1
'round' is not a member of 'std'

RatsM, 2022-08-04:
if you delete "std::", then it compiles.
it is also compiled with the string "displayCo2(result[0]);".
but esp 32 restarts endlessly

RatsM, 2022-08-04:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

psachs, 2022-08-05:
Hi @RatsM 

It looks like the ESP32  framework 1.0.6 you are using defines round in math.h and therefore throws the compilation error when using std::round.

From my experience depending on the Platform used either std::round is required since math.h is not defined or the other way around.

Does your ESP32 always restart when you call the `displayCO2` method or just in certain cases, e.g. when using without round?

Which verison of de BLE Gadget library are you using?

RatsM, 2022-08-09:
> Hi @RatsM
>
> It looks like the ESP32 framework 1.0.6 you are using defines round in math.h and therefore throws the compilation error when using std::round.
>
> From my experience depending on the [...]

the solution is elementary: instead of "displayCo2((uint16_t) round(result[0]));" you need to write displayCo2((uint16_t) round(result[0]));

BLE Gadget library v 1.0.0

psachs, 2022-10-14:
unfortunately currently it depends on the used Arduino environment and platform if math round or std round is available. We will consider to update the example to replace the rounding logic with something that is compatible with any system.

closed 2022-10-14 by psachs