Sensirion/arduino-ble-gadget #22: SwiftUI - receive BLE realtime values

sighmon, 2022-08-23:
Hi there,

Thanks so much for your sample code!

I'm writing a simple SwiftUI app for iOS to read the data from a SCD-41 sensor and send it onto an InfluxDB database.

I've incorporated your code into my [Arduino code](https://github.com/sighmon/co2_sensor_scd4x_esp32_http_server/blob/add/4-adafruit-qt-py-esp32-c3/co2_sensor_scd4x_esp32_http_server.ino#L78), and can read that using your MyAmbience app for iOS.

When I try and do the same using BLE in SwiftUI I'm able to request and download historic data, but can't seem to get realtime readings.

Here are the UUIDs I'm currently using:

```Swift
let co2Identifier = "50B30635-FC9C-57E6-A116-8FF87F780018"
// TOFIX: currently retrieves historic data
let co2MonitorServiceUUID = CBUUID(string: "00008000-b38d-4985-720e-0f993a68ee41")
let co2MonitorCharacteristicUUID = CBUUID(string: "00008004-b38d-4985-720e-0f993a68ee41")
```

What should I be using to get the realtime data?

psachs, 2022-08-24:
Hi @sighmon 

The live data is part of the BLE advertisement data that you receive through when you scan for devices.
This allows to get data without connecting to the gadget. We are currently writing some documentation on our BLE protocol that should help developers to write their own gadgets or Apps. This documentation will be published within the next few weeks. I will get you some more information on your specific use case later today

psachs, 2022-08-25:
If you want to read the CO2 value from a characteristic you have to use the following service
```
00007000-B38D-4985-720E-0F993A68EE41
```
CO2 ppm characteristic as uint16
```
00007001-B38D-4985-720E-0F993A68EE41
```

The service you are using is only for downloading and control historic data

sighmon, 2022-08-25:
@psachs Thanks for that information. I'm not seeing that `00007000 ` service or `00007001` characteristic. Here's my debug output:

```bash
BLE Peripheral found: <CBPeripheral: 0x2801a0500, identifier = 50B30635-FC9C-57E6-A116-8FF87F780018, name = S, mtu = 0, state = disconnected>, S (RSSI: -54)
BLE Connected to: <CBPeripheral: 0x2801a0500, identifier = 50B30635-FC9C-57E6-A116-8FF87F780018, name = S, mtu = 23, state = connected>
BLE Services: Optional([<CBService: 0x2825f1380, isPrimary = YES, UUID = 00008000-B38D-4985-720E-0F993A68EE41>])
BLE Characteristics: Optional([<CBCharacteristic: 0x2814a4d20, UUID = 00008002-B38D-4985-720E-0F993A68EE41, properties = 0x2, value = (null), notifying = NO>, <CBCharacteristic: 0x2814a4ea0, UUID = 00008001-B38D-4985-720E-0F993A68EE41, properties = 0xA, value = (null), notifying = NO>, <CBCharacteristic: 0x2814a4cc0, UUID = 00008004-B38D-4985-720E-0F993A68EE41, properties = 0x10, value = (null), notifying = NO>])
```

psachs, 2022-08-25:
@sighmon did you build your own gadget or do you have the "Sensirion SCD4x Gadget"?

For the DIY gadget, there is no service to retrieve live data but only live data through the advertisement data. This allows you to retrieve samples without connecting to the gadget, but you have to run a scan to get the advertisement samples.

sighmon, 2022-08-27:
@psachs I'm running your [example code](https://github.com/Sensirion/arduino-ble-gadget/blob/master/examples/Example8_SCD4x_BLE_Gadget_with_RHT/Example8_SCD4x_BLE_Gadget_with_RHT.ino) in my own [ESP32 code](https://github.com/sighmon/co2_sensor_scd4x_esp32_http_server/blob/add/4-adafruit-qt-py-esp32-c3/co2_sensor_scd4x_esp32_http_server.ino#L78).

What was your reasoning for removing the live data characteristic from your example code?

> For the DIY gadget, there is no service to retrieve live data but only live data through the advertisement data. This allows you to retrieve samples without connecting to the gadget, but you have to run a scan to get the advertisement samples.

I'll try that today, thanks.

sighmon, 2022-08-27:
@psachs I finally got the decoded values from advertisementData, but it didn't seem to be in the places you suggested. Instead I saw temperature starting at byte 6, humidity at byte 8, and co2 at byte 10. They also only decoded using the historic data decoder:

```swift
// For realtime data, read the advertisementData
let data = advertisementData["kCBAdvDataManufacturerData"] as! NSData
print("BLE ad data: \(String(describing: data))")

if !data.isEmpty {
    let temperature = data.subdata(with: NSMakeRange(6, 2)).withUnsafeBytes {
        $0.load(as: Int16.self)
    }
    let humidity = data.subdata(with: NSMakeRange(8, 2)).withUnsafeBytes {
        $0.load(as: Int16.self)
    }
    let co2 = data.subdata(with: NSMakeRange(10, 2)).withUnsafeBytes {
        $0.load(as: Int16.self)
    }
    print("BLE raw data: \(co2) \(temperature) \(humidity)")
    if co2 > 0 {
        self.co2Value = decodeCO2(co2: co2)
        self.temperatureValue = decodeHistoricTemperature(temperature: temperature)
        self.humidityValue = decodeHistoricHumidity(humidity: humidity)
        print("BLE decoded data: \(self.co2Value) \(self.temperatureValue) \(self.humidityValue)")
    }
}
```

Here's my debug output:

```swift
BLE ad data: {length = 18, bytes = 0xd506000a00d1f45e028eb004000000000000}
BLE raw data: 1200 24308 -29182
BLE decoded data: 1200 19.910353246356905 55.471122301060504

BLE ad data: {length = 18, bytes = 0xd506000a00d1165f418e3e04000000000000}
BLE raw data: 1086 24342 -29119
BLE decoded data: 1086 20.001144426642256 55.56725413900969
```

And my decoders:

```swift
func decodeHistoricTemperature(temperature: Int16) -> Double {
    return ((Double(temperature) / 65535) * 175) - 45
}

func decodeHistoricHumidity(humidity: Int16) -> Double {
    var humidity = ((Double(humidity) / 65535) * 100)
    if (humidity < 0) {
        humidity = 100 + humidity
    }
    return humidity
}

func decodeCO2(co2: Int16) -> Int {
    return Int(co2)
}
```

qfisch, 2022-09-01:
Hello @sighmon ,
We have made public a more extensive documentation on that subject.
You will find it [there](documents/Sensirion_BLE_communication_protocol.pdf)

Concerning the data itself starting at byte 4, it is correct. There are several bytes of "header" stored before.
Namely, 2 bytes of company identifier, 1 byte of sample type, and 1 byte of Sample advertisement type (see section 2 of the document)

sighmon, 2022-09-01:
@qfisch Great thank you!

closed 2022-09-01 by sighmon