Sensirion/arduino-ble-gadget #26: RAK AirQuality Node

ckuehnel, 2022-11-19:
Hi dev team
I build an air quality node base in RAKWireless Wisblock modules. The serial output is as follows:
[RAK_Output.pdf](https://github.com/Sensirion/arduino-ble-gadget/files/10047625/RAK_Output.pdf)
My program is based on your library examples.
[image]
Due to the use of SGP40, I have no CO2 value but a VOC index.  I used the T_RH_CO2_ALT datatype and hoped that ALT could be used for dimensionless values. But this was not possible. The screenshot shows the mismatch between CO2 in ppm and my index value.
What can I do in this case?
The next issue, battery voltage, is on your list already.
BR Claus

psachs, 2022-11-28:
You have to chose the appropriate data type so that MyAmbience knows what sensor signals it can expects.
Please find below the code you need to change:
This data type expects:
vocIndex in ppm (as float)
humidity in percent RH (as float)
temperature in degree Celsius (as float)

```c++
DataProvider provider(lib, DataType::T_RH_VOC);

provider.writeValueToCurrentSample(vocIndex), Unit::VOC);
provider.writeValueToCurrentSample(humidity, Unit::RH);
provider.writeValueToCurrentSample(temperature, Unit::T);
````

Please let me know if you need more help

ckuehnel, 2022-11-28:
I understand,  but VOC Index is dimensionless 0 to 500.I believe what you mean is eCO2 in ppm. Dr. Claus Kühnel - Talstr. 13b - CH-8852 AltendorfMobile +41-76-3960801

psachs, 2022-11-28:
Sorry about the confusion. Of course I meant VOC Index as unit less value between 0-500.
Seems my brain was not fully functional when I answered you 😉

ckuehnel, 2022-11-28:
And how can I handle such a value?Dr. Claus Kühnel - Talstr. 13b - CH-8852 AltendorfMobile +41-76-3960801

ckuehnel, 2022-12-08:
> Hi
> the VOC Index is a unit less value where 100 is an average and higher index references higher VOC levels.
> The VOC Index should mainly be used to detect events.
> Please find detailed information here:
> https://sensirion.com/media/documents/02232963/6294E043/Info_Note_VOC_Index.pdf

I know this all and use the Index.The question was how to handle VOC in the MyAmbience app.Dr. Claus Kühnel - Talstr. 13b - CH-8852 AltendorfMobile +41-76-3960801

ckuehnel, 2022-12-08:
As addition to my mail yesterday here is a screenshot.I used the CO2 slot to visualize iVOC. I know that not correct, but I did not find a slot to  visualize a number w/o a unit. A slot for battery voltage is missed too.BR Claus Dr. Claus Kühnel - Talstr. 13b - CH-8852 AltendorfMobile +41-76-3960801

https://sensirion.com/media/documents/02232963/6294E043/Info_Note_VOC_Index.pdf

psachs, 2023-01-11:
Please check my previous respond. You have to send the VOC index (0-500) using the correct sample type. I accidentally added a unit there. It is of course a unit less index and is also shown as such if you initialize the data provider with the correct sample type. If you send the VOC index using the CO2 example it will show up as CO2. 

```cpp
DataProvider provider(lib, DataType::T_RH_VOC);
```

The battery level interface is not released yet as we still have to fully support it in MyAmbience. This will be an interface to send battery in percentage and not in voltage.
You can however already use this feature if you take the library from the master branch. The MyAmbience release will take us until end of February, beginning of March

ckuehnel, 2023-01-15:
Now it works as expected. Thanks for the support.
Looking forward to battery support in March.
BR Claus

Dr. Claus Kühnel - Talstr. 13b - CH-8852 Altendorf
Mobile +41-76-3960801
-- Aus dem Rheinwerk-Verlag: Arduino - das umfassende Handbuch ISBN 978-3-8362-7345-9 [https://www.rheinwerk-verlag.de/arduino-das-umfassende-handbuch/] --
 

[Werbung_klein.jpg]

[8341a3e8-2799-41fa-8ecf-e6bd0ba87eca]

psachs, 2023-01-19:
thanks, good to hear that you got things working

closed 2023-01-19 by psachs