Sensirion/arduino-ble-gadget #30: How can we change the interval time ?

NetForces, 2023-05-02:
The MyAmbiance can set the interval time. If we change it to 1 minute via the app and the sensor reboots, it goes back to the default 10 minute.

stlljonas, 2023-05-04:
Hi,

to permanently change the interval time, you need to modify the private `_historyIntervalMilliSeconds` member in the `DataProvider` class. Concretely, you need to navigate to the folder where the library is installed (likely `~/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib`) and in `src/DataProvider.h` change the line 
```
uint64_t _historyIntervalMilliSeconds = 600000; // = 10 minutes
```
to
```
uint64_t _historyIntervalMilliSeconds = 60000; // = 1 minute
```
for an interal time of 1 minute.

Best regards and hope this helps!

NetForces, 2023-05-04:
Ok. My sensor has a menu where the user may change the intervals. So I need to dynamically adjust the `_historyIntervalMilliSeconds` and tell the user to adjust it in the app when they do change it.

I'll probably submit a pull request to have a public function to change `_historyIntervalMilliSeconds`. Unless you want to do it ;-)

NetForces, 2023-05-04:
Makes sense.

Also, I guess that the only valid `_historyIntervalMilliSeconds` value are the ones shown in the app dropdown selection ?

qfisch, 2024-03-26:
Hej @NetForces 
As far as I know an other interval than the ones proposed in _myAmbience_ could be selected.
myAmbience just provides a few presets.

Cheers
Q.

(Since this issue is very old, I will close it soon. Unless you come with new questions)

closed 2024-04-22 by qfisch