Sensirion/arduino-ble-gadget #33: Example2_SCD30_BLE_Gadget on a SparkFun Qwiic Pocket ESP32-C6

sighmon, 2024-03-13:
Hi there,

I've just bought a [Sparkfun ESP32-C6](https://www.sparkfun.com/products/22925) to use with an [Adafruit SCD30](https://www.adafruit.com/product/4867).

In the past I've had success building for an [Adafruit ESP32-C3](https://www.adafruit.com/product/5405), but am running into these errors:

* Arduino IDE 2.3.2
* Sensirion_Gadget_BLE_Arduino_Lib 1.2.0 [Example2_SCD30_BLE_Gadget.ino](https://github.com/Sensirion/arduino-ble-gadget/blob/master/examples/Example2_SCD30_BLE_Gadget/Example2_SCD30_BLE_Gadget.ino)
* NimBLE-Arduino 1.4.1
* ESP32 Espressif 3.0.0-alpha3

```
In file included from /Users/simos/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6/include/bt/include/esp32c6/include/esp_bt.h:17,
                 from /Users/simos/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLEDevice.h:46,
                 from /Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/NimBLELibraryWrapper.cpp:2:
/Users/simos/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6/include/bt/common/api/include/api/../../../../controller/esp32c6/esp_bt_cfg.h:19:10: fatal error: syscfg/syscfg.h: No such file or directory
   19 | #include "syscfg/syscfg.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

Guessing it was related to: https://github.com/h2zero/NimBLE-Arduino/issues/642

I tried using this branch: https://github.com/h2zero/NimBLE-Arduino/tree/esp32-c6-test

```
In file included from /Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/AdvertisementHeader.h:34,
                 from /Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/DataProvider.h:34,
                 from /Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/Sensirion_Gadget_BLE.h:35,
                 from /Users/simos/Documents/Arduino/Sensirion_Example2_SCD30_BLE_Gadget/Sensirion_Example2_SCD30_BLE_Gadget.ino:1:
/Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:40:10: error: 'string' in namespace 'std' does not name a type
   40 |     std::string getDataString() {
      |          ^~~~~~
/Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:1:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
  +++ |+#include <string>
    1 | /*

exit status 1

Compilation error: exit status 1
```

So then I added `#include <string>` to `Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h`

```
/Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/DataProvider.cpp: In member function 'void DataProvider::writeValueToCurrentSample(float, SignalType)':
/Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/DataProvider.cpp:42:50: error: 'std::map<SignalType, SampleSlot>::mapped_type' {aka 'struct SampleSlot'} has no member named 'converterFunction'
   42 |         _sampleConfig.sampleSlots.at(signalType).converterFunction;
      |                                                  ^~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

qfisch, 2024-03-18:
Hello @sighmon ,

What version of `Sensirion_UPT_Core` do you have installed ?
In `0.3.0` some changes have been made and BLE gadget is not compatible yet.

Q.

qfisch, 2024-03-24:
You can get UPT Core 0.2.0 from the [repository](https://github.com/Sensirion/arduino-upt-core) as a zip and load it into the Arduino IDE using "Load .Zip library"
That should allow you to continue with your project until we update this library

Q.

qfisch, 2024-03-26:
Hej @sighmon 

in the end the changes were already done in code but were never released.
I created the version **1.3.0** which include the necessary changes to be compatible with the _latest "UPT Core"_

Q.

closed 2024-03-26 by qfisch

sighmon, 2024-04-04:
@qfisch I'm still seeing this error when trying to compile with:

* Sensirion Core `0.6.0`
* Sensirion UPT Core `0.3.0`
* Sensirion Gadget BLE Arduino Lib `v1.3.0`:

```
/Users/simos/Documents/Arduino/libraries/arduino-ble-gadget/src/ByteArray.h:40:10: error: 'string' in namespace 'std' does not name a type
   40 |     std::string getDataString() {
      |          ^~~~~~
```

qfisch, 2024-04-04:
> @qfisch I'm still seeing this error when trying to compile with:
>
>     * Sensirion Core `0.6.0`
>
>     * Sensirion UPT Core `0.3.0`
>
>     * Sensirion Gadget BLE Arduino Lib `v1.3.0`:
>
>
> ``` [...]

Hej @sighmon 

Yes this release only solved the dependency issue with UPT Core. 
We would need to look further at the usage of string within the lib...
For now adding `#include <string>` seems to be a reasonable workaround