Sensirion/arduino-ble-gadget #34: Usage of std::string without import

qfisch, 2024-04-04:
```
In file included from /Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/AdvertisementHeader.h:34,
                 from /Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/DataProvider.h:34,
                 from /Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/Sensirion_Gadget_BLE.h:35,
                 from /Users/simos/Documents/Arduino/Sensirion_Example2_SCD30_BLE_Gadget/Sensirion_Example2_SCD30_BLE_Gadget.ino:1:
/Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:40:10: error: 'string' in namespace 'std' does not name a type
   40 |     std::string getDataString() {
      |          ^~~~~~
/Users/simos/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:1:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
  +++ |+#include <string>
    1 | /*

exit status 1

Compilation error: exit status 1
```

Found by @sighmon on [issue 33](https://github.com/Sensirion/arduino-ble-gadget/issues/33)
Quick fix is to add `#include <string>`

sighmon, 2024-04-07:
@qfisch Note this fix only works with Sensirion/arduino-upt-core@919ef8b and not `0.3.0`

qfisch, 2024-04-29:
The issue was fixed with version 1.3.1
@sighmon as far as we could see the import indeed fixed the issue, but we also fixed an import (lib name)
Feel free to come back to us if you encounter further issue

Cheers
Q

hallinen, 2024-05-17:
From your previous post you wrote..."but we also fixed an import (lib name)".  Should I have deleted the Sensirion_UPT_Core and replaced it with the renamed version?  When I uploaded 1.3.1, I did it without the dependencies since it gave me an error when I tried to upload the dependencies at the same time.

qfisch, 2024-05-21:
Hello @hallinen,

Sorry about the trouble, we have a hard time to fix those import issues since we often can not reproduce them.
We updated the dependencies because UPT core was renamed (to remove the underscores). the latest version 1.3.1 should not have any import failures. What was the error message  for import ?

Thank you
Q.

hallinen, 2024-05-22:
Thank you for the reply.

I initially had trouble installing 1.3.1 because it said something about having multiple versions of Core, I didn't copy the message, I'm sorry. I fixed it by deleting the Core folder and the older versions of Gadget_BLE.  I just uninstalled 1.3.1 and deleted the core folder again and reinstalled 1.3.1 with dependencies.  It didn't give me any trouble.  I didn't delete the NimbleBLE file. I tried to compile example 8, the SCD4x sketch, and got the same error.

In file included from /Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/AdvertisementHeader.h:34,
                 from /Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/DataProvider.h:34,
                 from /Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/Sensirion_Gadget_BLE.h:35,
                 from /private/var/folders/tc/p1d_fw296cb18b8wnf6rdm280000gn/T/.arduinoIDE-unsaved2024421-18525-hez030.813c6/Example8_SCD4x_BLE_Gadget_with_RHT/Example8_SCD4x_BLE_Gadget_with_RHT.ino:4:
/Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:41:10: error: 'string' in namespace 'std' does not name a type
   41 |     std::string getDataString() {
      |          ^~~~~~
/Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:1:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
  +++ |+#include <string>
    1 | /*

exit status 1

Compilation error: exit status 1

I tried to compile the other examples, same error. I got the same error for this sketch that just consists of  #include "Sensirion_Gadget_BLE.h"

#include "Sensirion_Gadget_BLE.h"
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
 

/Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/AdvertisementHeader.h:34,
                 from /Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/DataProvider.h:34,
                 from /Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/Sensirion_Gadget_BLE.h:35,
                 from /private/var/folders/tc/p1d_fw296cb18b8wnf6rdm280000gn/T/.arduinoIDE-unsaved2024421-18525-1b43f4o.5snrf/sketch_may21a/sketch_may21a.ino:1:
/Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:41:10: error: 'string' in namespace 'std' does not name a type
   41 |     std::string getDataString() {
      |          ^~~~~~
/Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:1:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
  +++ |+#include <string>
    1 | /*

exit status 1

Compilation error: exit status 1

qfisch, 2024-05-22:
Hello @hallinen 
I am trying to reproduce your setup. Could you specify for which board you are compiling ?

Thanks
Q

hallinen, 2024-05-22:
I've tried using ESP32 Dev Module, ESP32c3 Dev Module and AirM2m-core-exp32c3, same error...   Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:41:10: error: 'string' in namespace 'std' does not name a type
   41 |     std::string getDataString() {
      |          ^~~~~~
/Users/dianehallinen/Documents/Arduino/libraries/Sensirion_Gadget_BLE_Arduino_Lib/src/ByteArray.h:1:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
  +++ |+#include <string>
    1 | /*

  Prior to the last update I was making sensors with the small esp32c3 boards and didn't have any problem compiling or uploading them.  They are wonderful little sensors!

qfisch, 2024-06-07:
Hello @hallinen,
A colleague managed to reproduce your error.
He managed to compile by adding `#include <string>` (not string.h) in the following files:
- arduino-ble-gadget: ByteArray.h
- upt-core: BLEProtocol.h
- upt-core: Measurement.h

Could you try that ? If that solved the issue we will release fixed versions 

Cheers
Q

hallinen, 2024-06-07:
I was able to compile the code! Thank you. It works faster!

I wasn't able to do a forced calibration. I can do it on the Gadget I purchased from you, but I don't see that option in the app for my DIY. I'm wondering if this code needs to be changed.

// GadgetBle workflow
static int64_t lastMeasurementTimeMs = 0;
static int measurementIntervalMs = 5000;
NimBLELibraryWrapper lib;
// initialize DataProvider with SCD FRC BLE Servie enabled
DataProvider provider(lib, DataType::T_RH_CO2, false, false, true);
char errorMessage[256];

Thanks again!

hallinen, 2024-06-09:
Update: I loaded the sketch onto a different brand of ESP32C3 and the calibration works just fine.  Bug is squashed!  Thank you so much.

qfisch, 2024-06-10:
Hello @hallinen 

That is great to hear. Sorry for the trouble
And enjoy using your device

psachs, 2024-06-19:
Issue is fixed with
Arduino BLE Gadget Version 1.3.2
Sensirion UPT Core Version 0.3.2

closed 2024-06-19 by psachs