Sensirion/arduino-ble-gadget #37: Sporadic reboots

NetForces, 2024-05-17:
I have been having sporadic reboots when using this. It might have to do with the fact that my device sleeps, but I am not sure.

When a reboot occurs, I get the following in the console:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
```

I will attempt to provide the minimal code to reproduce this situation in the next couple of days but if anyone had this issue and know a fix it would be appreciated.

LeonieFierz, 2024-05-21:
Hi @NetForces 

we are not aware of any issues with reboots when using the arduino-ble-gadget library. 
Depending on your setup and the sensor used, it might be that the power supply is not sufficient. Which sensor do you have connected?

Best, Léonie

NetForces, 2024-05-21:
It's the SCD-41.

I will revalidate, but if I remove the part of my code that does the light sleep, the sporadic reboots go away. Unfortunately, light sleep is required in my application as it runs off a small battery.

LeonieFierz, 2024-05-21:
As you write that you are running from a small battery, please note that the SCD41 draws up to 200mA during measurement.

Let us know how it goes and we are happy to help if you have further questions regarding the sensor.

LeonieFierz, 2024-05-22:
The SCD41 peak power during measurement is 200mA, the low power mode just does the measurement less often and puts the sensor to sleep in between the measurements.
 
We have not worked with the light sleep mode on Arduino so I do not have any experience with it, but if you wake the board up and do the measurement I could imagine that the battery voltage could drop due to the power draw from startup and measurement. To see what is happening it would make sense to measure the voltage.

NetForces, 2024-08-05:
So I had some more time to spend on this and it is not a power issue. 

I am plugged in the USB port (for charging, power and serial output and I removed the SCD-41 and I get the same sporadic reboots just doing mock reading (returning random values).

I will try to provide a minimal firmware code that reproduces this in the next days.

LeonieFierz, 2024-08-28:
Hi @NetForces 
did you find time to create the minimal firmware code?

LeonieFierz, 2024-12-05:
@NetForces  Please Re-Open the issue or post a new one in case you have and update or further questions.

closed 2024-12-05 by LeonieFierz