Sensirion/arduino-ble-gadget #38: Download interruption Issue when Fetching Data from ESP32 in MyAmbiance App

melkati, 2024-05-18:
Hi guys.

I encountered a download interruption issue when attempting to fetch data from the ESP32 device using the "Fetch from Gadget" option in the MyAmbiance app. I compiled the official basic example (Example1) to ensure independence from my project, installing it from Arduino IDE. Despite these efforts, the issue persists, indicating a potential problem within the library or MyAmbiance.

Steps to Reproduce:

Compile and upload the official basic example (Example1).
Open the MyAmbiance app.
Navigate to the "Fetch from Gadget" option.
Observe the interruption when attempting to fetch data from the ESP32 device.
Expected Behavior:
The data should be fetched seamlessly from the ESP32 device without any interruptions or discrepancies.

Additional Information:

Tested with MyAmbiance 3.3.1.
Installed from Arduino IDE for isolation from my project (developed with VS Code and PlatformIO).
This issue seems to occur regardless of the specific project setup, pointing towards a potential issue within the library or MyAmbiance.
Any insights or assistance in resolving this issue would be greatly appreciated.

Please, find more information, along with some user reports, in this issue at my project: 
https://github.com/melkati/CO2-Gadget/issues/216

Also, I'm not sure if it's related to this other issue I already reported some days ago:
https://github.com/Sensirion/arduino-ble-gadget/issues/36

LeonieFierz, 2024-05-23:
Hi @melkati 

thank you for reporting and investigating on the issue. I did look into it and found two bugs in the MyAmbience app leading to this behavior. MyAmbience only considers a download as finished if all samples could be retrieved from the gadget. As BLE has no guarantee that all packets arrive, it happens that not all data packs sent by the gadget are received in MyAmbience. This leads to the "Download interrupted" error shown. Second thing is that the number of actually downloaded samples is bigger than the number of expected samples (e.g. 455/102). MyAmbience only requests to download the samples not already downloaded by writing the "Requested Samples" characteristic. The arduino-ble-lib does not implement this characteristic and MyAmbience ignores that the characteristic is not available, leading to the fact that the number expected is actually lower than what the Gadget sends. 

We will address both issues in the next MyAmbience release. 

In Fact, in the background MyAmbience saves all the samples that were received and thus, even though the message states that the download failed and the shown numbers are odd the data is stored. 

To avoid the odd numbers you could implement the "Requested Samples" characteristic (please refer to the BLE specification https://github.com/Sensirion/arduino-ble-gadget/blob/master/documents/00-Sensirion_BLE_communication_protocol.pdf, Chapter 1.3 "Data Logger Service"). This would also improve download time, as it would allow MyAmbience to only download the new samples from the gadget.

melkati, 2024-05-24:
Hi @LeonieFierz,

Thank you for your detailed response and for identifying the bugs in the MyAmbiance app. I appreciate the clarification and the background on how the download process works and the issues causing the interruptions.

I'm glad to hear that the next MyAmbiance release will address these problems. I will look into implementing the "Requested Samples" characteristic as described in the BLE specification to improve the download process and avoid the odd numbers issue. However, I'm concerned that implementing this characteristic might be quite complicated, as I have no experience working with Bluetooth beyond what this library provides. Nevertheless, I will give it a try.

Thank you again for your assistance and for providing a clear path forward.

closed 2024-06-19 by psachs