Sensirion/arduino-ble-gadget #44: Slow speed data download and minor issue

danielbernalb, 2024-10-17:
Hi, 
My main issue is the very slow data download speed in the MyAmbience app. Using the same Android phone, it sometimes takes minutes to download, for example, 200 data points, while other times it only takes few seconds. I haven’t been able to identify the cause of this inconsistency. I would appreciate some guidance on resolving this issue.

And I’ve noticed a minor issue with the MyAmbience app displaying the Device ID in an unhelpful way.

Example in the APP:

Device id: 00:C1 (device 1)
Device id: 00:c1 (device 2)

However, in the code I’ve written, the ID is displayed correctly:

  Serial.print(F("Sensirion Provider Lib initialized with deviceId = "));
  Serial.println(provider.getDeviceIdString());

Sensirion Provider Lib initialized with deviceId = c1:a2

Sensirion Provider Lib initialized with deviceId = c1:d3

Thanks!

LeonieFierz, 2024-10-22:
Hello @danielbernalb 

1. Slow Data Download

First thing to check is that you always keep your phone near the gadget while downloading and also having a lot of other BLE devices around might slow down the download.

Further, we recently added support in MyAmbience and arduino-ble-gadget to avoid missing datapoints when downloading data from the gadget. This means that whenever we detect that data was lost we re-initialize the data download. The support for it was added on the master, but was not yet released. Which version do you use, latest master or version 1.3.2?

2. Device Id 
thank you for reporting the inconsistency with the Device ID. We fixed that with https://github.com/Sensirion/arduino-ble-gadget/pull/45 and will release it with the next version.

Best, Léonie

danielbernalb, 2024-10-28:
Hello Leonie. Thanks for your response.

Testing my SEN55-4-0, SPS30 and SCD40 I found differences:

1. If I configure SEN55 with:
    _DataProvider Provider (lib, datatype::T_RH_VOC_NOX_PM25)_
    generates the error:
    __assert failed: void ByteArray::writeByte(uint8_t, size_t) [with unsigned int SIZE = 20; uint8_t = unsigned character; size_t = unsigned int] ByteArray.h:56 (position < SIZE)_
    always.

2. If I configure SEN54 with:
    _dataProviderProvider(lib, dataType::T_RH_VOC_PM25_V2);_
    It works well but very slow.

3.  If I configure the SEN50 or SPS30 with:
    _dataProviderProvider(lib, dataType::PM10_PM25_PM40_PM100);_
    It works well and is fast.

4. SCD40:
    _dataProviderProvider(lib, dataType::T_RH_CO2_ALT);_
    It works well and is fast.

Due to the problem I change the DataProvider for SEN55 to T_RH_VOC_PM25_V2 and I have the same performance (good but slow) and if I change to PM10_PM25_PM40_PM100 it works fine and fast.

I think it's a problem with the performance of the DataProvider.

Thank you

LeonieFierz, 2024-11-08:
Hello @danielbernalb 
thank you for investigating further. It is not clear to my why the different data types should have a influence on the data speed. I will try to reproduce and come back when I have more infos.
Best, Léonie

LeonieFierz, 2024-11-26:
Deair @danielbernalb 

I had some time to investigate regarding the data download

1. There is a Bug in the definition of the download type used for T_RH_VOC_NOX_PM25 which we need to solve on the Arduino and MyAmbience side. So for now please use e.g. T_RH_VOC_PM25_V2 for a Gadget with SEN55.

2. I did test the data download using the T_RH_VOC_PM25_V2 as data type and could not observe a slow download. Do you use the examples provided in this repo? If so, please try to remove the delay(1000) in the measure_and_report() function which is actually not necessary as the timing is done in the loop() and this just blocks processing, including the handling of the download, for too long. If you have your own code, please make sure that the provider.handleDownload(); get called every 5ms or so and the measurement is not blocking for longer than needed.

I hope this helps to resolve the issue with fast/slow downloads.

Best, Léonie

LeonieFierz, 2025-02-10:
I consider that the above fix and advice did resolve the problem. If not, please re-open this issue.

closed 2025-02-10 by LeonieFierz