Sensirion/arduino-ble-gadget #56: Compile problem

lxmatic, 2025-05-23:
Hello,

i have a problem when compiling(Arduino IDE) the example(**Example15_SEN66_BLE_Gadget**) for use with MyAmbience. I did as described in tutorial --> documents/**15-SEN66_BLE_Gadget_Tutorial.md** and getting this error:

_c:\Users\MaticZorko\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src\NimBLELibraryWrapper.cpp: In member function 'virtual void NimBLELibraryWrapper::init()':
c:\Users\MaticZorko\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src\NimBLELibraryWrapper.cpp:120:32: error: 'class NimBLEAdvertising' has no member named 'setMinPreferred'
  120 |     _data->pNimBLEAdvertising->setMinPreferred(0x06);
      |                                ^~~~~~~~~~~~~~~
c:\Users\MaticZorko\Documents\Arduino\libraries\Sensirion_Gadget_BLE_Arduino_Lib\src\NimBLELibraryWrapper.cpp:121:32: error: 'class NimBLEAdvertising' has no member named 'setMaxPreferred'
  121 |     _data->pNimBLEAdvertising->setMaxPreferred(0x12);
      |                                ^~~~~~~~~~~~~~~
exit status 1

Compilation error: exit status 1_

Kindly asking for your assistance.

BR, Matic

lxmatic, 2025-06-03:
Hello @qfisch ,

first of all thank you for a quick response.
I´m using ESPRESSIF(ESP32-WROOM-32D), latest IDE. When I2C example loaded and run i get normal data via serial monitor... only when BLE gadget launched i get this error from above.

Regards,
Matic

sdmueller, 2025-06-24:
Hi @lxmatic 

`setMinPreferred` and `setMaxPreferred` methods have been removed in the 2.x Versions of `NimBLE-Arduino` library.
Can you check what version of the library is used when you launch it on the gadget?
We enforce a version <2.0.0 via the library.properties file but maybe you are running with a newer one on the gadget for some reason?

closed 2025-08-01 by lxmatic